Physics-Dude/Phys-Index-EyetrackVR-HW #1: right eye print

rawpie2, 2023-12-14:
I only see left eye in the .stl files, where is the one to print for the right eye? thanks.

Physics-Dude, 2023-12-14:
You have to mirror the model in your slicer to print the right eye ring. (In Prusa Slicer for example, right click the part > Mirror > Mirror X).  
This simplifies version control and encourages you to print one ring at a time which reduces stringing.

Physics-Dude, 2023-12-14:
I created [ETVR ValveIndex Cam-LED Mount Rings/Index-EyetrackVRv4 v18 combined L and R plate.stl](https://github.com/Physics-Dude/Phys-Index-EyetrackVR-HW/blob/d3e680a5e1266e084980f29253d3a4de4e84feea/ETVR%20ValveIndex%20Cam-LED%20Mount%20Rings/Index-EyetrackVRv4%20v18%20combined%20L%20and%20R%20plate.stl) with both mirrored copies of the rings if desired. Just be careful on stringy printers.

closed 2023-12-14 by Physics-Dude