Physics-Dude/Phys-Index-EyetrackVR-HW #2: [Model question] "Inverted" gum stick model

tompointexe, 2024-01-24:
Hi, with the usb hub i got, mounting and solderiung would result in the boards beeing on the wrong side, do you have a model with is "inverted" from the curent one, see pics below
[image]
[image]

Thanks a lot

Physics-Dude, 2024-01-24:
That is weird. I've never came across that model of USB hub before. Do you have a link to the product pace so we can avoid that in the future? It looks like it has an LED in the middle which might be a good indicator for future product hunts. 

A mirrored version wont work since all your solder pads will still effectively be on the wrong side of the USB hub's PCB. You might be better off soldering your thinnest wire (28-32AWG) to all the hub pads you'll need, then sliding it into the gum stick you already printed so those pads are effectively facing downwards and hidden.

Alternatively, here is a version of the hum stick that has the spacer between the ESPs and USB hub PCB removed. This should give you more room for the soldered pads and wire to snake out from under the usb hub's pcb and into the cavity for later connection to the ESPs. I'd suggest a strip of tape on the ESPs though to prevent any shorts if you dont trust your soldering.

[No-Spacer alternate - XIAO-ESP32S3 ETVR v4 gum stick index mount v24.stl](https://github.com/Physics-Dude/Phys-Index-EyetrackVR-HW/blob/main/Gum%20Stick%20Dongle%20Custom%20USB%20Hub%20Case/No-Spacer%20alternate%20-%20XIAO-ESP32S3%20ETVR%20v4%20gum%20stick%20index%20mount%20v24.stl)

Keep us updated so we can mention it in the doc.

tompointexe, 2024-01-24:
Yup inverted is not realy the term, it would need to have the pcb insert and usb port where the front facing usb port is, and put that where the curent hub pcb insert is
[image]

The hub i've used is this one, it's the only one i could find on amazon.fr / Europe, i've not searched aliexpress tho
https://www.amazon.fr/dp/B0CLY9KZ68?

Physics-Dude, 2024-01-24:
The way I understand it, flipping the whole gum stick model on X or Y, (or inserting the hub from the other direction as per your initial edit) will not expose the HUB's solder pads since those are referenced to the orientation of the main USB port that plugs into the headset. 

I would personally buy the correct hub from aliexpress. Re-mounting the main port sounds too invasive considering its molded into plastic from the factory.

The No-Spacer alternate body I linked previously does give you about 1mm of room to solder and route wires from 'under' the usb hub pcb to anywhere else in the cavity of the gum stick (as long as you solder the wires before inserting the hub into the main body of the gum stick with the corrected orientation). If you got the time waiting for a new hub, I'd still try this so at least you'd have a working spare if it doesn't work out. 

I'll put a note in the readme to avoid the similar-looking USB hubs with LED indicators.