Physics-Dude/Phys-Index-EyetrackVR-HW #5: confused about the soldering

fleufy, 2024-10-15:
Hi! It's my first time doing this sort of thing, I've never soldered or done any type of electronics before, sorry if I'm just being stupid!
I just started the soldering steps and wired everything (shown in the picture) but I messed something up.
I thought I had to solder both of the ESPs in the same places but your photo shows the wires of the ETVR connected to the soldering pads that (I thought) were reserved for the 2nd ESP.
Now I'm not sure what to do with my wires, if you could help me, it would mean the world!
[image]

Physics-Dude, 2024-10-15:
Hi, your build is looking good and that is a great question. The two wires from the ETVE LED driver board are supposed to share any 5v and Gnd pad on the USB hub. I should have made this more clear in the guide, but here is a picture that may help.

[image]

If it works for you, I may include this pic in the main write-up. Let us know if you have any more questions.

fleufy, 2024-10-15:
Thank you very much! That absolutely solves my issue. Have a great day!

closed 2024-10-15 by fleufy