Physics-Dude/Phys-Index-EyetrackVR-HW #6: USB Hub Requirement confusing cased me to get the wrong one

AAMBIENTT, 2024-11-05:
Would like to poing out the one in the image of the guide is one you DONT want us to buy. Thats has the LED and too short for all of the adapters. Please use an image like this one to tell us what to look for 
Wrong: https://www.amazon.com/Aluminium-Splitter-Expander-Console-Keyborad/dp/B0BJZ753D9
Right: https://www.amazon.com/YaimhSound-Extender-Splitter-Extension-Keyborad/dp/B0BXJ21DXZ

Physics-Dude, 2024-11-05:
You are right, the listing has changed. I will update the writeup now.
(i will also strike out the old one just in case it is a temporary amazon change.)

closed 2024-11-05 by Physics-Dude

AAMBIENTT, 2024-11-06:
could you please update the image as well, as that model in the image is the one i purchased and it was very wrong.

Physics-Dude, 2024-11-06:
done! there is also a gum stick frame variant that should allow for the flipped hub design as well.