Physics-Dude/Phys-Index-EyetrackVR-HW #7: ETVR v4 PCB Wiring diagram missing

AAMBIENTT, 2024-11-07:
the ETVR v4 PCB Wiring diagram is missing. Currently at the step of wiring everything together and i cannot figure out how to wire in the ETVR v4 PCB.

Physics-Dude, 2024-11-07:
I unfortunately do not have any pictures of the underside of my vr board, but in summery:

1. Solder two wires to the "VIN" + and - pins
2. Feed them through the little hole that goes from the ETVR pcb area, to the USB hub area (you can see these purple/black wires in the writeup)
3. Solder these respective + and - wires to any 5v and GND pads on the USB hub. You can share the same pins any ESPs is using for power.

let me know if this helps and i can include it in the write up.

AAMBIENTT, 2024-11-07:
Perfect, will try once i receive the correct hub in the mail, i will take as many images as possible for you here

AAMBIENTT, 2024-12-16:
Ok, so im having an issue with the et r module of when the power switch is set to on, i get an error in windows mentioning theres a power surge and it exceeds the power limit of the hub. Any fix?

Physics-Dude, 2024-12-16:
That means there is a short circuit or solder bridge somewhere between VUSB and Gnd. You should inspect all your soldered connections and make sure everything is soldered to the right places. Also make sure no solder or pieces of wire is touching places it shouldn't.

AAMBIENTT, 2024-12-16:
Ok, so i think i got it, but im not sure how to test to make sure that the LEDs and etvr board are working. Is there any way to test these to see if they appear?

AAMBIENTT, 2024-12-16:
> Look at the IR LEDs with your phone camera. you should see a pink
> dot/light in each led. There are some examples in the EyeTrackVR Discord
> server.

I dont seem to see any leds.

Physics-Dude, 2024-12-16:
Check out the ETVR Discord server. The search bar can give you a trove of past fixes and pictures. and lots of people there can help.

https://discord.gg/kkXYbVykZX

closed 2024-12-16 by Physics-Dude