Physics-Dude/Phys-Index-EyetrackVR-HW #8: Mouth Tracker Wiring diagram needed

AAMBIENTT, 2024-11-29:
I saw you mentioned we needed resitors. i have one here but i do not see where you have yours hidden here. is it truly needed?

Physics-Dude, 2024-11-29:
A resistor is indeed needed in most cases, but the value is dependent on the particular IR LEDs you chose. 

With two series 1.6v, 20mA IR LEDs I used, a 100 ohm series resistor was required. 

In my case, I hid the resistor under the camera ribbon between the two LEDs on the face camera arm. You can also solder it to the 5v vbus pin, just make sure it doesnt short out to other parts of the ESP.

if this helps, let me know. I was thinking about adding a FAQ to the write up. 👍

AAMBIENTT, 2024-12-16:
I was reading the guide while wrapping this up and ended up removing one of the pads from one of my esps (data+) so i was just going to plug the usb into the c port of the esp, but will that still deliver power to the 5v rail on the board? Will that provide power even if its powered through the c port

Physics-Dude, 2024-12-16:
Yes, you can connect any of the same pads from a USB-C breakout connector to where they would go on the USB hub. You can use any combination of pads and plug wires (i.e. use VUSB and Gnd where they normally go, and D- and D+ from the usb connector you add).

AAMBIENTT, 2024-12-17:
Ok so i think this will be our final interaction! Everything is done and wired, however, the mouth LEDs appear to be super dim. I have it wired as such
VBUS>100r>3mm 940nm Emitter #1 Anode->3mm 940nm Emitter #1 Cathode>3mm 940nm Emitter #2 Anode->3mm 940nm Emitter #2 Cathode>GND

Is this right? As mentioned, it is incredibly dim.

Physics-Dude, 2024-12-17:
Your circuit is correct, but due to the variety of IR LEDs out there, its a guessing game for finding the resistor value. If your source of LEDs tells you the max volts/current value on the product page, use that in a "LED Series Resistor Calculator".

If the image from the camera is still dark while mounted to your headset and pointed at your face, it is safe to assume you can lower your resistor value slightly to increase brightness. Check the Project Babble docs or Discord server for references on how dark/bright the image should be.  https://discord.gg/c99B6vB2sX

I would try a 75 ohm resistor next, then a 50 ohm after that if it is still too dark.

closed 2024-12-17 by Physics-Dude