Physics-Dude/Phys-Index-EyetrackVR-HW #10: Face tracker mount; LED safety/supply question

Datgek, 2025-01-12:
Asking this because I'm not very mathematically skilled and have 0 experience when messing around with LEDs.
(I know I'm probably just over complicating this on my end, but I'd rather ask an unnecessary question than risk blinding myself.)
(And if the answer is just, "go on amazon, it's most likely fine" that'd be good too lol.)

The ETVR website, under "LED Safety", lists the equation for a light's radiance as "mW/cm2/sr" I believe that I understand the equation correctly, but I'm having trouble figuring out how to get the mW as needed for it. As well as figuring out the "sr" portion.

By my understanding, getting the mW listed in the equation is done with LEDs supplied voltage, minus the forward voltage per each LED wired in series(?), then divided by the resistor's ohm rating.
The only issue I'm having (and it might not even be an issue), is that for any LED I put into a resistor calculator, the outputted resistance seems lower than expected, such as the addicore 5mm IR LEDs, the resistance when calculated with 2 LEDs is 18 ohm, which just seems way too low to me...

I. . . have been flipping through so many tabs to double check my numbers and logic while typing this that I forgot what my original point was...  but just in general, what would I need to look out for so that I don't cause damage to anything, and/or am I just over complicating this like crazy?

Physics-Dude, 2025-01-13:
Don't confuse an LED's light irradiance (mw/cm^2) with its power consumption (mW), these are very different numbers. 

Power consumption (mW) is defined by the LEDs rated forward voltage (typically between 1.2v to 1.6v for IR leds) multiplied by the forward current (typically 30-50mA). Your supply voltage is always going to be about 5v here. 

Resistor = (Supply Voltage / LED Voltage) / (LED Current)

Most 3mm and 5mm IR LEDs I seen have a 1.2v forward voltage, and 30mA current limit. You'll usually need two of these.
Two in series are equivalent to a 2.4v led at 30mA. This pair would require a 100 ohm resistor if supplied by 5v.

Adafruit sells a nice efficient set of 5mm LEDs that are 1.6v and 100mA. You dont need this to be full power, it will be too bright on the camera, so its best to 'dim' it by pretending its a 50mA LED or so.

As for safety;

LED safety is only considered when non-visible light is shining directly into the eye balls as there is existing research concerning that. LEDs of this scale shining on skin are far less concerning (some peer reviewed sources even indicate it is beneficial to an extent, but thats beyond this scope). 

For the LEDs in the eye-tracking rings, I can only suggest the official ETVR kits in full.

For selecting the LEDs in the face tracker, due to the variance in LED sources and user choice, the best thing you can do is test with a high value resistor (i.e. 220 ohm) then work your way down in standard increments. (I.e. 150, 100, 75, 47, 22) looking at your camera feed while everything is mounted and see if your face is too light/dark. nothing in this range at 5v is considered dangerous to bare skin.

Worst case, if you go too low on the resistor and make the led shine brighter than it was designed to, it'll be hot to touch and you'll burn it out.

Datgek, 2025-01-13:
Ok, I'm realizing now that I was operating under the pretense that, if I were to try and run an LED at a current lower then it's listed forward current, it simply wouldn't work, this explanation though was better than 90% of the info I found online.

As for the Adafruit LED, I did come across it while looking, but alongside the aforementioned pretense (thinking that it'd only function at 100+mA), it was also listed as 940nm, so I wasn't sure if it wouldn't show as well on a 850nm camera.

All in all though, thank you for the (_actually_ easy to understand) explanation.

closed 2025-01-13 by Datgek