Physics-Dude/Phys-Index-EyetrackVR-HW #13: Wire Routing

MrVampy99, 2025-03-11:
I am unsure on where to solder the power connections for the ETVR v4 PCB as I am new to electronics. Any help would be nice

Physics-Dude, 2025-03-11:
On the bottom of the v4 PCB, the VIN + pad connects to any positive 5v pin on the usb hub. This is the same spot any ESP's VUSB pin connects to.

The VIN - pin on the bottom of the ETVR v4 PCB connects to any Gnd pin on the USB hub. You can share the same pad that any Gnd pin any ESP connects to. 

[image]

closed 2025-03-12 by MrVampy99

Marokarr, 2025-03-13:
Can the 5V VIN+ pad that is already supplying power to the ESP32 also be used at the same time to power the v4 PCB?

Physics-Dude, 2025-03-13:
yes