Physics-Dude/Phys-Index-EyetrackVR-HW #15: What do we do if there are no 200mm OV2640 modules available anywhere at all?

TheSpiderDungeon, 2025-03-15:
Does it have to be _EXACTLY_ 160 degrees? Can it be 75mm long? Because I've checked 12 sites and none of them have both 200mm _and_ 160 degree FOV.

Physics-Dude, 2025-03-15:
Many have resorted to getting the 24-pin Extension ribbon and Extension connector when the 200mm cameras are unavailable. Both available on Aliexpress and Adafruit.

https://docs.eyetrackvr.dev/how_to_build/part_list

GarnetSunset, 2025-04-15:
For anyone lost in the sauce like me:
https://www.aliexpress.us/item/3256802853835121.html
Color: 75MM-160 850nm

and 
https://www.aliexpress.us/item/2255799835842411.html
200mm A , Insert Type: 24p 0.5mm pitch

Physics-Dude, 2025-04-16:
Don't forget the female-female extension connector so you can connect the camera ribbon to the extension ribbon:

https://www.aliexpress.us/item/3256804096715690.html
Click: 24 pin

GarnetSunset, 2025-04-16:
Bought, thank you!

EnmaAisha, 2025-12-30:
The downside is that the extensions are wider, so they can't fit through the standard frunk, unless you snip it open or make it wider.