Physics-Dude/Phys-Index-EyetrackVR-HW #16: What's the Difference in steps if we use the V4 mini Solder Required LED Kit?

xlDarkDragonlx, 2025-03-21:
Im really new to modding vr but im going to be following the guide with my step dad and i was wondering what the difference would be if we used V4 mini Solder Required LED Kit instead of the no solder cause it is currently sold out, and since i have you here i also pre ordered a babble facial tracker will this block the mount for that too or will i have to print my own mount for it

Physics-Dude, 2025-03-21:
The solder-required ETVR v4 kit just means you have to supply your own 28-32AWG solid core wire and solder to the pads on the respective leds and power PCB. The ETVR discord can answer any more questions. 

It looks like the "replacement wires" are in stock on the ETVR store. I would buy a couple of those to make life easier.

[image]

Physics-Dude, 2025-03-21:
I am 99% sure the solder-required kit still has the connectors installed on the PCBs. The pads you would solder to are on the underside of the PCB where the power input pads are.