Physics-Dude/Phys-Index-EyetrackVR-HW #20: Curious about compatibility with prescription inserts

TrebleSketch, 2025-05-07:
Thinking about giving this a crack before Furality, was wondering if this project is compatible with the prescription insets from [VROptician](https://vroptician.com/prescription-lens-inserts/valve-index)?

If not, what would further modifications would be reququired to get this working if at all? (Somehow hooking onto the prescriotion lens frame itself?)

Physics-Dude, 2025-05-07:
Plenty of people have used prescription insets with ETVR on index. However I have only passively followed the development. For that, you should consult the official ETVR Discord server. Files are usually shared and iterated upon there. I'm sure there's a few public repos with comfortable camera + led + inset mounts by now they could link to.

Physics-Dude, 2025-05-22:
the 3mf files are designed to open only in Autodesk Fusion 360. You can use it for free as a hobbyist.