Physics-Dude/Phys-Index-EyetrackVR-HW #22: Question about D- and + pad.

Jinxatron, 2025-07-10:
Hello, I received the parts for the kit; however, I've encountered some issues. The Digital pads on the back of the 32S3 didn't seem to stick when soldering, and when I finally achieved a good bond, they ripped. Now I assume the board is toast at this point, but I figured I'd ask if there is anywhere you could reroute those wires?

Fixing the trace is a no-go since I simply don't have the equipment, nor can I see where the signals go on the board.

Physics-Dude, 2025-07-11:
You might be able to use a simple USB-C (2.0) breakout with some modification to the gumstick case. 
But if solder isn't sticking to the pads it could be:

1. Your solder does not have flux/rozen built in (aka rozen core solder). Use flux in addition to it, or get rozen core 60/40 solder.  
2. Soldering iron is too hot/cold. Try 300ºC for 60/40 leaded solder.

Jinxatron, 2025-07-11:
You know, that is a very simple yet effective way to solve it. I will look into it, at first glance it just looks like I need to add some channels on the side for the cables. 

and I deffo think I was running the solder to hot, after some research it seems delamination happened.

Jinxatron, 2025-07-11:
Yeupp. Impatience is expensive in the long run. Thank you for all the tips, it should come in handy. And thank you for designing this, much preferable over replacing the entire front.